domesticatedviking/skinnydip #2: Slic3r PE 1.42.0-beta1 - Error

Robagon3D, 2019-04-11:
Hello I have tried to configure skinnydip per your guide, however I get an error when trying to export g-code.  I get slic3r error code 2.  I looked in the skinnydip.py folder and no log file has been generated.  It seems like it is not even running the script.

I am using windows 10 and have python 2.7.16 32bit installed.
 
[image]
[image]
[image]

Thanks!

domesticatedviking, 2019-04-11:
Is there a skinnydip.log file in the same directory as skinnydip.py?   If so, could you please attach it?  I know there was a bug in Slic3r's error handling when I was initially writing the script that caused it to fail silently when the script failed.    Alternately, you could run the skinnydip.py script from the command prompt eg
``\path\to\python.exe \path\to\skinnydip.py \path\to\myfile.gcode `` 
and post the output.   Thanks!

Robagon3D, 2019-04-12:
I got it to work!  Looks like the spaces in my file paths were the issue.

closed 2019-04-12 by Robagon3D

domesticatedviking, 2019-04-12:
> I got it to work! Looks like the spaces in my file paths were the issue. Thanks for your help with this and your work on skinnydip, can't wait to try it out on one of my stringy filaments.

You're welcome!   I've recently learned that Slic3r will refuse to parse post processing scripts that have more than one parameter (to protect against profiles being a vector for doing nasty things to people's systems).    Not sure whether the spaces in the filepaths triggers that issue, but I'll need to write a batch file for windows users because a lot of you are having trouble with this.