domesticatedviking/skinnydip #4: toolchange_temp not working

Whuzz, 2019-05-21:
First off, great tool! It has reduced my MMU Load Failures by 80% on the first attempts in using it, which is massive. After days and days, and hours and hours of trying every other setting, this finally gives me hope for my MMU2S. 

Some info: 
MK3S, MMU2S
Slic3r 1.41.3+win64
Printing combination of PLA and PVA, using skinnydip only on the PVA (for now). 

One thing I notice is that the toolchange_temp doesn't seem to be working, or maybe I just don't understand how it's supposed to work.

I use two filaments. One PVA at 190C and one PLA at 205C. In the filmanent settings for the PVA I have:

`M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{else}30{endif}; Filament gcode
; SKINNYDIP CONFIGURATION START
; material_type PVA
; material_name JupiterPVA
; insertion_speed 2000 
; extraction_speed 4000
; insertion_pause 0 
; insertion_distance 46 
; removal_pause 0
; toolchange_temp 190
; beep_on_dip off
; beep_on_temp off
; SKINNYDIP CONFIGURATION END
M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{elsif nozzle_diameter[0]==0.6}15{else}30{endif} ; Filament gcode`

I think this should make the PVA stay at 190C during toolchange, right? However, what happens is that as soon as the PVA move to the purgetower for purging and unloading, the nozzle starts heating up to 205C (for the PLA). By the time unloading starts the nozzle temp reads ~202C. 

I'd like the nozzle to stay at 190C and only start heating up once the PLA gets loaded. How do I achieve that?

domesticatedviking, 2019-05-21:
The first thing to verify is that the configuration is being picked up properly by the script.  I've seen one instance where the regex fails to pick up the settings and some of the extruders don't have their settings applied.   Check the skinnydip header (open the processed gcode in the text editor) to confirm that the tool number printing PVA is in the active tools list.   If it isn't there, chances are this is the configuration bug I've been trying to track down, in which case it would be very helpful to me if you could send both your pre-processed and post-processed gcode files so that I can figure out why the regex isn't working properly.

Whuzz, 2019-05-22:
I know skinnydip itself is working. Behavior of the first extruder (PVA) is definitely (positively) affected. Here's my header from the post processed code:

```
; SKINNYDIP THREAD REDUCTION v1.0.3 beta
; https://github.com/domesticatedviking/skinnydip
; Postprocessing completed on Tue May 21 18:41:59 2019
;               File Processed:I:\Bubo_skinnydip.gcode
; Note that editing the values below will have no effect on your
; Skinnydip settings.  To change parameters you must reslice.

;         Configured extruders: ['T0']
;             Toolchange temps: [190]
;            Insertion lengths: [46]
;      Auto insertion distance: 46.5
;       Total # of toolchanges: 71
;                   Dips added: 69
;       Toolchange temps added: 35
;   Tools beeping on skinnydip: None
; Tools beeping on temp change: None

; generated by Slic3r Prusa Edition 1.41.3+win64 on 2019-05-21 at 18:41:54
```

T0 I assume is the first extruder, so that seems correct. When I get home tonight, I'll provide the full pre- and post-process codes. Should I send them by email? If so, where?

domesticatedviking, 2019-05-22:
Yes, your script is being run properly.  My preference would be that you would post your gcode files here via dropbox or google drive.  If that isn't possible due to intellectual property issues, let me know.

domesticatedviking, 2019-05-22:
Ah, I also just noticed some duplication in your filament gcode.   Try deleting the bottom M900 line

```
M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{else}30{endif}; Filament gcode
; SKINNYDIP CONFIGURATION START
; material_type PVA
; material_name JupiterPVA
; insertion_speed 2000 
; extraction_speed 4000
; insertion_pause 0 
; insertion_distance 46 
; removal_pause 0
; toolchange_temp 190
; beep_on_dip off
; beep_on_temp off
; SKINNYDIP CONFIGURATION END
**M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{elsif nozzle_diameter[0]==0.6}15{else}30{endif} ; Filament gcode**
```

Whuzz, 2019-05-23:
Pre-Process: https://drive.google.com/file/d/1EWn50u1kBOT8bX-rLGxnwxYNdJG0apRi/view?usp=sharing
Post-process: https://drive.google.com/file/d/1Gd15IWu_pR3qxA_y9UXlmugaTjHBFHGS/view?usp=sharing

This is with that duplicated line though. I will remove that from the settings. The lines are slightly different though, the bottom one is there by default. I suppose that doesn't matter then? (I have the standard MK3 nozzle)

domesticatedviking, 2019-05-23:
The lines are slightly different though, the bottom one is there by default. I suppose that doesn't matter then? (I have the standard MK3 nozzle)

Any unexpected text in the filament start gcode section could potentially cause the regex to miss the settings.  Ideally the script will be able to deal with that, but it might not be there yet.   I would suggest replacing the top M900 line with the updated one ie. the one on the bottom.

domesticatedviking, 2019-05-23:
Also, I should clarify:  By "non-processed" gcode, what I need is something that has been sliced with the skinnydip parameters present in the filament profile, but which has not been run through the skinnydip script.   Sorry I didn't make that clear.       I need the settings to be present so that I can see what the regexes are matching or failing to match.

Whuzz, 2019-05-24:
Ah ok. So I should just remove the pre-processing line (X:\python27\xxxxx X:\xxxx\skinnydip.py) from the print settings, leave the filament settings with the skinnydip params, then generate the gcode, right?

domesticatedviking, 2019-05-24:
> Ah ok. So I should just remove the pre-processing line (X:\python27\xxxxx X:\xxxx\skinnydip.py) from the print settings, leave the filament settings with the skinnydip params, then generate the gcode, right?

Yes exactly.

Whuzz, 2019-05-26:
Ok, here they are. 
The PLA filaments (extruder 5) and PVA (extruder 1) both have Skinnydip params with different temp settings. pre-process is without the Skinnydip script in the output of the print options. post-process is with the script. 
Weird thing is... the post-process script is smaller in size? I'd expect it to be larger? Hope I didn't screw it up again...

Pre: https://drive.google.com/file/d/1VpTp0v5WU29Nl0TyaSlNa-OhbZlE-5mK/view?usp=sharing
Post: https://drive.google.com/file/d/12VTcwrgVvjsDDLS36wOPwaEqyhNiN5wX/view?usp=sharing

domesticatedviking, 2019-05-26:
OK, so I have had a chance to look at this.
Here is a section of your post-processed gcode.   Everything here looks fine to me.

```
; *****************************************
M109 R190 ;***SKINNYDIP Waiting for T0 toolchange temp: 190
; *****************************************
G1 E-15.0000 F6000
G1 E-24.5000 F1200
G1 E-7.0000 F600
G1 E-3.5000 F360
M104 S205
G1 Y136.860
G1 X170.750 E20.0000 F488
G1 X221.180 E-20.0000 F326
;*****SKINNYDIP THREAD REDUCTION*****************
; Tool(T0), PVA/JupiterPVA
G1 E46 F2000  ;move stringy tip into melt zone
G1 E-46 F4000  ;extract clean tip from melt zone
;************************************************
G1 E-35.0000 F2000
G1 Y137.000 F2400
G4 S0
T4
```

1. the tool moves to the wipe tower, 
2. The toolhead pauses until it hits the toolchange temperature,  (the M109 Rnnn command)
3. Filament is yanked out of the hotend (a series of G1 E-nnnnnn commands)
4. The hotend immediately starts heating to the print temperature (NOT the toolchange temperature) of the next filament that will be used.   It doesn't pause while it does that in order to save time, this also gives the print temperature a chance to stabilize before the toolhead returns to the model.
5. Cooling moves are performed as requested in advanced filament settings.
6. A skinnydip sequence will happen if one has been requested.
7. The filament is extracted from the heatsink and retracted up to the MMU parking position.

I'm going to close this issue now but feel free to comment or reopen it if I've misunderstood what you're experiencing.

closed 2019-05-26 by domesticatedviking

Whuzz, 2019-05-28:
Hmm, perhaps I hadn't understood the toolchange temp correctly. I expected the hotend to stay at the toolchange_temp until *after* the skinnydip sequence. 

Let's say I print my PVA at 190C and my PLA at 215C and I am toolchanging from PVA to PLA. With the current events it means that the PVA is being "skinnydipped" into a hotend that is well above 190C, because it's already heating up to 215C. If I understand your above post, that is the expected behavior? 

Isn't that going to cause stringing of the PVA again?