domesticatedviking/skinnydip #5: 0 dips added

jettoblack, 2019-05-22:
I'm using the TZB MMU2 firmware and a modified BMG-like extruder so my setup certainly isn't standard. I'm also using PrusaSlicer 2.0 final. This problem is likely due to my non-standard setup but I'm reporting it anyway in case this is useful.

After post-processing with v1.0.3 beta, temperature settings appear to be applied at every tool change correctly, but it reports that 0 dips are added. This may be due to the modified extruder settings needed for TZB firmware (different load distance etc.), or maybe something that changed in PrusaSlicer since 1.42, I'm not sure.

G-code and log file: https://www.dropbox.com/s/i5b41dm4ot7mo5c/skinnydip%200%20dips.zip?dl=0

domesticatedviking, 2019-05-22:
Thank you for this very complete bug report.   The use of TZB firmware shouldn't be a problem unless there are surprises in your filament start gcode or toolchange gcode that break the regular expressions that are used to detect the settings.   I've had good luck post processing files sliced with PrusaSlicer so far.

Could you please do me a favour and try reslicing that  model with Printer Settings > Firmware > Supports Remaining Times unchecked?   That feature litters the generated gcode with M73 commands,  which may be causing the regex to fail.    It would save me some time correcting this issue if I could narrow it down to this reason.

jettoblack, 2019-05-23:
Here you go, turning off remaining times didn't seem to change anything.

https://www.dropbox.com/s/8pay7tlzcmsy5ip/skinnydip%200%20dips%20no%20m73.zip?dl=0

domesticatedviking, 2019-05-23:
Thanks for that.  Doesn't seem to be the M73 causing it.  That makes this bug a lot more interesting.   Let me know if you have any other models with similar behaviour, this is a new one for me.

domesticatedviking, 2019-05-26:
Hey @jettoblack I'm looking at this issue and realizing that it would be a lot easier to track down if I had a copy of what your gcode file looked like before you post processed it. Would you be able to send me a copy of the gcode file, sliced with your skinnydip filament settings present, but which hasn't been run through the skinnydip script?

jettoblack, 2019-05-29:
Here you go, before and after plus log: https://www.dropbox.com/s/sm0dc8s57c1n232/skinny%20before%20after.zip?dl=0

This is not 100% identical to the g-code from before since I didn't save that plate.

domesticatedviking, 2019-05-29:
> This is not 100% identical to the g-code from before since I didn't save that plate.
As long as the same error is present, I can use this.   Thank you again.

domesticatedviking, 2019-05-30:
Thank you very much for your feedback.  Please keep documenting any issues you find and I'll do my best to keep fixing them.

closed 2019-06-06 by domesticatedviking