domesticatedviking/skinnydip #6: Suggestion: Add "cooling tube pause"

germanyxx3, 2019-05-23:
Heyo,
so using this script is awesome! I struggled 3 weeks before I discovered skinnydip to get these damn tips right. It really reduces stringing 
I'm printing a lot with ABS and I noticed that one additional setting could help to reduce stringing even more than toolchange_temp.
Because the whole hotend gets of course even warmer the longer it prints it has an impact to increase stringing over the duration of the print.

Now to describe my idea - after the cooling moves are made (maybe even before that?) and the filament got pulled out of the hotend but before skinnydip is executed (with toolchange_temp enabled or not) the filament should be pulled out close to the bondtech gears - this is the coldest possible region for the filament cool down. ABS takes quite some time to cool down and get somewhat stiff again. Now with using "cooling tube pause" instead of "toolchange_temp" for ABS it not only cools it down faster but also makes it in a shorter peroid of time more stiff. 
Being stiff also has the advantage that the Filament tip is not deforming much when it gets pushed to the hotend - so the possibility to make the filament tip bigger than 1.75mm should be noticalbe reduced.
After the configured time, for example 5 seconds have passed the skinnydip operation should be executed as usual.

I'd like to add this but I barely have a clue about programming...

Any questions? Feel free to ask

cheers

domesticatedviking, 2019-05-23:
Thanks for your feedback!   The cooling moves built into slic3r already accomplish this.   If you need more cooling prior to skinnydip, you should add more cooling moves or slow them down.    I've done all my testing of this script printing with ABS, and the only tip deformation due to temperature I have seen has been when I unloaded the filament too quickly after the skinnydip move and the bondtech gears crushed the tip.   This is why I added the option to pause prior to unload.

closed 2019-05-23 by domesticatedviking

germanyxx3, 2019-05-23:
The problem with the cooling moves are that when they are executed the ABS is still **not** stiff and that's why it can deform. Even at low speeds and I don't wana crank it all the way down to 0.5mm/s or slower. ABS should be not moved when it cools down?

You also mentioned in facebook that you have these special tube inserts with a smaller diameter. A lot of people don't and probably will never have until prusa takes action here. When you already have a smaller inner diameter of course the Filament will deform less or basically nothing. Or did you do your tests on a machine without these special inserts?
When I look at what you posted in the prusa community group (https://www.facebook.com/photo.php?fbid=10155865734536601&set=p.10155865734536601&type=3&theater) that only confirms my guess that you had the special inserts.
Why? because no matter what I did most of my filament tips have this small stubborn "addition"
Take a look at the pictures
I already changed my ptfe tube once to a capricorn xs series with 1.9mm inner diameter but that didn't seem to have so much of an impact
[image]
[image]

domesticatedviking, 2019-05-23:
I use custom Torlon heatsink inserts that are about 1.81mm I.D. to keep tip diameters down.   It is not a good idea to use tubing that is any bigger than 1.85mm in the heatsink.  The issue with PTFE in the heat sink is that PTFE compresses too easily, resulting in larger tips.   

These images show evidence of previous colors lingering in the heatbreak assembly, which could be the result of a gap between your heatsink PTFE and your heatbreak, or possibly excess retraction.  Are you using a collet clip on the top of your heatsink?    It is critical that there not be anything for the hot filament to snag on when it is being removed.  The condition of the bottom of your PTFE is important, and it is also important that no material is gathering there.

I suspect your insertion_distance may be set too high on skinnydip as well, which may be causing the entire tip to be remelted during the dip cycle.

germanyxx3, 2019-05-23:
I am using a collet clip on my heatsink, yes I quadrouple checked it when I changed my tube to capricorn xs and when I first installed it I aslo quadrouple checked it. There is no play. the issue simply is because the tube has a too big diameter and as we know expands even more when being heated up.
Now here's an important thing:
When the ABS is being quickly pulled out. Just pulled out and NOT even a slight bit being pushed back these additions are not attached to the tip but as the filament is being pushed back down they slightly fuse to the tip. I'm saying slightly because it is easy to remove them even by hand (not completely though). Sometimes they fuse stronger and sometimes not so strong. 
The filament melts before it reaches the nozzle. Two times I removed the nozzle out of the hotend and manually pushed ABS through. (Hotent temp was 250c) It was so soft when I was able to touch it with pliers and it was no job at all to strech it centimeters long.

My point here is, when the ABS tip would be cooled down near the bondtech gears for a defined amount of time it may can stand the chance that there will fuse no extra filament to the tip and therefor not having these "additions"
I would really like to have a ptfe tube (or any tube) that has a smaller diameter and does not deform but the custom ones from David Shealy are currently sould out and he has no chance to meet the demand of all MMU2 users.
This issue should be hardware sided solved but this is taking so much longer than just trying out a software sided idea. I think you can agree on that.
When was the last time you saw an ABS filament tip with the standard ptfe tube?
I bet they won't look much different than mine. Also keep in mind that since I recived the MMU2s about three weeks ago I was adjusting and tuning **every day**. I have more tips than on this picture and on every print I hade these stubborn additions on the filament tip. Sometimes less and sometimes more but they were always present and caused loading issues. I think you can imagine why.

Would it be a lot of effort to try "cooling tube pause" on one print?
Just pause the tip like 5mm before the bontech gears for 5 - 10 seconds - then execute the cooling moves or skinnydip and then see if the Tip looks different.

domesticatedviking, 2019-05-24:
The spec for the MMU2 is 1.85mm I.D. PTFE in the heatsink, (1.80mm is actually better, if only it stayed that size.)  Your substitution of 1.90mm is likely making your problems with tip diameter considerably worse.  If I see evidence that people with fully stock extruders are having similar problems to yours I could be persuaded to add this feature, but that probably won't happen anytime soon.

germanyxx3, 2019-05-24:
I said that since I changed my ptfe tube to capricorn xs there is no visible difference so I guess that even though the capricorn tube is slightly bigger it probably stays around 1.9mm whereas the standart ptfe tube quickly expands to 1.9mm inner diameter...
Sadly I found no tube that has an inner diameter of 1.75, 1.8 or similar that I could buy...
I guess waiting for a reaction from prusa will take even longer. I am also willing to donate some money if this would speed up the process. I am that convinced, yes. And I am frustrated.
I keep tweaking with your script for sure nevertheless. Still I think cooling tube pause could be the best
compromise