domesticatedviking/skinnydip #9: Significantly less dips than expected

delawarehunter, 2019-06-07:
I noticed during an MMU2 "Torture" test, that there weren't many dips being performed.  After looking at the gcode, I noticed 102 tool changes but only 21 dips added.  I've attached the gcode with the dips added to this issue.  Renamed the extension to .txt so I could upload the file.  

I also have the 3 color lizard model that is showing 121 tool changes but only 81 dips added.  I'll attach 2 version of the gcode (renamed to .txt) to this issue as well.  

[test.txt](https://github.com/domesticatedviking/skinnydip/files/3264087/test.txt)
[test_dipped.txt](https://github.com/domesticatedviking/skinnydip/files/3264088/test_dipped.txt)

[MMU2_3c_test.txt](https://github.com/domesticatedviking/skinnydip/files/3264084/MMU2_3c_test.txt)

delawarehunter, 2019-06-07:
Also, ignore the message about the toolchange temp... I fatfingered a 3 at the end for one of my filament types.  I fixed that but i'm still seeing the discrepancy in # of dips.

domesticatedviking, 2019-06-07:
Here are the settings from your unprocessed gcode.  One significant issue is that you have specified a toolchange temperature of **2003C.**

```
M900 K30; Filament gcode
; SKINNYDIP CONFIGURATION START
; material_type PETG
; material_name ABasics PETG
; insertion_speed 4000 
; extraction_speed 4000
; insertion_pause 0 
; insertion_distance 41 
; removal_pause 0
; toolchange_temp 2003
; beep_on_dip off
; beep_on_temp off
; SKINNYDIP CONFIGURATION END
```

In your processed gcode, we see that the script caught that these settings were out of bounds, and set them to the maximum temperature the printer firmware permits:  
```
; SKINNYDIP THREAD REDUCTION v1.0.5 beta
; https://github.com/domesticatedviking/skinnydip
; Postprocessing completed on Thu Jun  6 21:48:01 2019
;               File Processed:D:\3Dprinting\Stock from Prusa\samples\Lizard\test.gcode
; Note that editing the values below will have no effect on your
; Skinnydip settings.  To change parameters you must reslice.

;         Configured extruders: ['T0', 'T1', 'T2']
;             Toolchange temps: [295, 295, 295]
;          Insertion distances: [41, 41, 41]
;      Auto insertion distance: 46.5
;       Total # of toolchanges: 120
;                   Dips added: 81
;       Toolchange temps added: 279
;   Tools beeping on skinnydip: None
; Tools beeping on temp change: None

; SOME PARAMETERS WERE OUT OF SAFE RANGES AND WERE CORRECTED!
;   Maximum setting toolchange_temp for T0 enforced: 295
;   Maximum setting toolchange_temp for T1 enforced: 295
;   Maximum setting toolchange_temp for T2 enforced: 295

```
It appears that when you sliced the model, you only had 2 of the 5 filament channels configured with skinnydip parameters in their filament gcode.    This could either be because you forgot to put the parameters into the filament profiles that you used, or because you didn't select those filament profiles when you sliced this model.     These are the profiles that you used:

```
"Amazon Basics PETG.ini (Billy Edit)";
"Prusament PLA MMU2";
"Amazon Basics PETG.ini (Billy Edit)";
"ABasics Black PLA 1.75mm";
"ABasics Black PLA 1.75mm"
```

There does appear to be a bug in either the settings regex or the the range-checking function. T1 should not have been assigned a toolchange temperature at all.    In fact, the bug here is not that there were fewer dips than expected, but more dips! 
I will investigate further.

domesticatedviking, 2019-06-07:
I am thankful you caught the fat finger issue before attempting to print that file.
Here are the profiles used in your third gcode file:

```
"Amazon Basics PETG.ini (Billy Edit)";
"eSun PET Orange";
"Amazon Basics PETG.ini (Billy Edit)";
"ABasics Black PLA 1.75mm";
"ABasics Black PLA 1.75mm"
```
Here is the filament gcode for those profiles (reformatted a little to see the settings for each extruder)

```
; start_filament_gcode = 
"M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{else}30{endif};

 Filament gcode\n; SKINNYDIP CONFIGURATION START\n; material_type PLA\n; material_name my PLA Spool\n; insertion_speed 2000 \n; extraction_speed 4000\n; insertion_pause 0 \n; insertion_distance 41 \n; removal_pause 0\n; toolchange_temp off\n; beep_on_dip off\n; beep_on_temp off\n; SKINNYDIP CONFIGURATION END"

"M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{else}30{endif}; Filament gcode\n; SKINNYDIP CONFIGURATION START\n; material_type PLA\n; material_name my PLA Spool\n; insertion_speed 2000 \n; extraction_speed 4000\n; insertion_pause 0 \n; insertion_distance 41 \n; removal_pause 0\n; toolchange_temp off\n; beep_on_dip off\n; beep_on_temp off\n; SKINNYDIP CONFIGURATION END";

"M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{else}30{endif}; Filament gcode\n; SKINNYDIP CONFIGURATION START\n; material_type PLA\n; material_name my PLA Spool\n; insertion_speed 2000 \n; extraction_speed 4000\n; insertion_pause 0 \n; insertion_distance 41 \n; removal_pause 0\n; toolchange_temp off\n; beep_on_dip off\n; beep_on_temp off\n; SKINNYDIP CONFIGURATION END";

"M900 K{if printer_preset==\"Original Prusa i3 MK2 MultiMaterial\"}200{elsif printer_preset==\"Original Prusa i3 MK2 MM Single Mode\"}200{else}30{endif};

 Filament gcode";"M900 K{if printer_preset==\"Original Prusa i3 MK2 MultiMaterial\"}200{elsif printer_preset==\"Original Prusa i3 MK2 MM Single Mode\"}200{else}30{endif}; Filament gcode"

```
This means you have 3 / 5 filament channels configured to run dips.  Was that what you were expecting?

delawarehunter, 2019-06-07:
ah crud... that's odd because I was sure that I wasn't using that Prusament filament setting... I should have caught that.  Sorry for wasting your time...

delawarehunter, 2019-06-07:
Here is a link to an updated gcode with all filament using the skinnydip start gcode:

https://1drv.ms/u/s!AtdaMwOhL3FFlV9JO_BWeBWvKsCE?e=xwG6WM

domesticatedviking, 2019-06-07:
OK, so that shows you have the configuration in the right place for all 5 materials.    Unfortunately I need the unprocessed version of this file to see what the regular expressions are missing.

delawarehunter, 2019-06-07:
Oh sorry, my mistake.  Here are 2 new files that I just created this morning:

Dipped: https://1drv.ms/u/s!AtdaMwOhL3FFlWHjnmzzWitrEvPF?e=mopZc9
Undipped: https://1drv.ms/u/s!AtdaMwOhL3FFlWJiGYgqJgyk0wgQ?e=8GUMQf

Just to add to the information.  This is the 3 color lizard stl's from Prusa.

domesticatedviking, 2019-06-07:
Fascinating.  I think you've got this configured properly, but there is configuration data missing in places.  This could be related to some optimizing of the gcode that I haven't seen before and which the script can't manage.    Thank you for posting those two files.  If I could also get the skinnydip.log file (generated in the same folder as your skinnydip.py file is located) I'd appreciate it -- would help me rule out some other issue.

These kinds of surprises are inevitable given how the post processor works. The version of skinnydip I have integrated into PrusaSlicer doesn't have to work nearly as hard to get consistent results.   I really hope I can get someone to adopt it in a mainstream build.

delawarehunter, 2019-06-07:
I'm a little torn.  I'm happy that I could help you with optimizing the code, but I'm sorry I found an issue for you to have to look at, haha.    Either way, I fully understand it being a work in progress and didn't have expectations that it would be 100% full proof.  I can tell you that the times I have seen it dip, it works flawlessly.  I don't believe I am doing anything out of the ordinary with optimizing the code, if there is anything I can provide from PrusaSlicer that will help you diagnose, let me know. 

I will get the log uploaded when I get home later tonight (it'll be around 9:30 or so US Eastern).  I almost uploaded it to my Onedrive before I left this morning... If my wife is home, I may ask her to set up my Teamviewer account so I can get it to you in a few minutes.  

I was in the process of trying to Compile the PrusaSlicer build that you posted before I went to bed last night, but one of the steps left me with 7 errors that I wasn't able to troubleshoot.  Hoping to look at them tonight.  I'm no where near knowledgeable enough to code, just dangerous enough to be able to follow directions and build the Windows version based on the instructions that you have and maybe diagnose issues.  If I get the Slicer compiled and working, I'll try it with the same settings and let you know if I see any problems.

delawarehunter, 2019-06-07:
Here is the log:  https://1drv.ms/u/s!AtdaMwOhL3FFlWOLJ8g5Lz-pGhM3?e=eMk8bf

domesticatedviking, 2019-06-07:
It's a delight to have the support of people who can communicate clearly when I'm on a bug hunt.   Re compiling the prusaslicer build -- I'm in exactly the same boat as you are when it comes to getting the build environments set up for all the the platforms people use.   I have some people trying to build a windows version for me today.  I'll make sure you get a link to some binaries  if / when that happens. 
The PP script is never going to be my preferred method of adding these routines -- I was quite angry when it took me less than a day to add them to PrusaSlicer (written in C++ which is a language I have no prior experience with).   I probably spent a solid week and a half on the PP script, and I'm pretty comfortable with Python.

delawarehunter, 2019-06-07:
I'd like to think I am competent at the "easy" languages like VB.  I am ok at troubleshooting something else wrote in Python, but even that is a little bit of a struggle.  Then again, I've never really put any significant amount of time into try to teach myself.

domesticatedviking, 2019-06-07:
> I'd like to think I am competent at the "easy" languages like VB. I am ok at troubleshooting something else wrote in Python, but even that is a little bit of a struggle. Then again, I've never really put any significant amount of time into try to teach myself.
I think I see a lot of parallels between your experience and my own.  If I'm right about that, I bet you'd love the free courses on codecademy.com.    A few years ago I spent 3-4 months picking up the fundamentals of Ruby, Javascript, and Python there.  I *love* their interactive lesson format.

delawarehunter, 2019-06-08:
I'll have to check that out.  I did something similar a while back, some sort of free coding website, but I can't remember what it was called.  

Just out of curiosity, your script is supposed to dip on every toolchange right?

delawarehunter, 2019-06-08:
To make things interesting... I only used 2 colors instead of 3.  Now I am getting 88 toolchanges and 86 dips.  Here are all of the files:

Dipped GCode: https://1drv.ms/u/s!AtdaMwOhL3FFlWQVgpgYHqbdw0ol?e=Nn3HAg
Undipped GCode: https://1drv.ms/u/s!AtdaMwOhL3FFlWYsd94N6_eJhAXC?e=9Clw90
Dipped LOG File: https://1drv.ms/u/s!AtdaMwOhL3FFlWV9fOLDUZd3AfBw?e=96cib8

delawarehunter, 2019-06-08:
Sorry, messed up on the last files so I deleted the comment...  let me try it again.  I used the same model, but only 2 colors instead of 3.. 88 ToolChanges, 86 dips:

Dipped GCode: https://1drv.ms/u/s!AtdaMwOhL3FFlWnSt0ldBtWYDTM5?e=C1e5i0
Undipped GCode: https://1drv.ms/u/s!AtdaMwOhL3FFlWiaXnkF_61MPidE?e=9FeDno
Skinnydip Log: https://1drv.ms/u/s!AtdaMwOhL3FFlWcVrOvd-1tJnwDl?e=DhLlX4

delawarehunter, 2019-06-09:
Ok.  I've been doing a lot of testing over the last half an hour... and I think I might have narrowed down something... I just don't know what it means.  Let me see if I can give the basics...

I am trying to run a 3 color print.  I am using 2 different brands of filament (eSun and Amazon Basics) it's all PETG.  If I slice the model with all ABasics filaments or all eSun filaments... I get 155 tool changes with 153 dips.  Here are the files:

dipped:  https://1drv.ms/u/s!AtdaMwOhL3FFlWsM7uVsLoGe88WX?e=NCOAiT
undipped: https://1drv.ms/u/s!AtdaMwOhL3FFlWwr2mhcxunQMvuB?e=E4ADgZ
log: https://1drv.ms/u/s!AtdaMwOhL3FFlWpniShpwaRogvdC?e=UF4aIh

I get the same exact results if I run it with all eSun.  I can upload files if you want.  

If I change Filament 1 to eSun and leave Filament 2 and 3 to ABasics, I get 155 tool changes and 4 (!!) dips... Here are the files:

dipped: https://1drv.ms/u/s!AtdaMwOhL3FFlXSUpGYQj4Sp0Jmo
undipped: https://1drv.ms/u/s!AtdaMwOhL3FFlXMtQ-6lhkF8--BK
log: https://1drv.ms/u/s!AtdaMwOhL3FFlW2RRtM2QFNyBb3C

As before, I get the same results if I flip the filaments (eSun in 1, ABasics in 2 and 3)

If I set 1 to ABasics, 2 to eSun, 3 to ABasics, I get 155 tool changes and 48 dips.  Just to let you know, this is the same filament setup that I used on the GCode that started this issue.  The only difference it that I am assigning the #2 filament to a different part in the lizard.  Previously, I was assigning filament 2 (eSun) to only the eyes on the lizard.  In order to keep this "experiment" similar under each circumstance I am not changing the part -> filament assignments.  Regardless, here are the updated files:

dipped: https://1drv.ms/u/s!AtdaMwOhL3FFlXbOJrazTbRtQ__d
undipped: https://1drv.ms/u/s!AtdaMwOhL3FFlXXdqCILpzjq42Gu
log: https://1drv.ms/u/s!AtdaMwOhL3FFlXIuQooFXALKbByu

I get the same exact results if I flip the filaments (eSun in 1 and 3, ABasics in 2)

Finally, if I set ABasics to Filament 1 and 2, and eSun to Filament 3, I get 155 tool changes and 107 tool changes.  Here are the files:

dipped: https://1drv.ms/u/s!AtdaMwOhL3FFlXnUKqhDaKMaqhjg
undipped: https://1drv.ms/u/s!AtdaMwOhL3FFlXgisrvVX_BOLEHo
log: https://1drv.ms/u/s!AtdaMwOhL3FFlXe5G65nglnqJyDB

I get the same exact results if I flip the filaments (eSun in 1 and 2, ABasics in 3)

Right now, those 3 filaments are the only ones that I have set up in PrusaSlicer with the SkinnyDip parameters.  Hopefully this helps figure out what might be causing it.  

I'm not having any luck with compiling your SkinnyDip Integrated version of PrusaSlicer... I plan to keep working on that later tonight.

domesticatedviking, 2019-06-09:
Great code samples.   Should be very useful.   I'm working on setting up build environments for the integrated version, won't be able to look at this issue again for a couple of days.    In the meantime, check your filament gcode for any unprintable characters.   I don't see anything obviously wrong in your configuration.

Ideally the script will cause a dip every time a filament is being extracted from the hotend.   Currently it doesn't do that if you are printing test strips at the beginning of a print, nor does it dip on the final extraction.   Those conditions may be added later, but clearly there are a few bugs in the core configuration that are causing transient and mysterious issues that are more important right now.   It is very normal for the number of hits for toolchange temperatures to be much higher than the number of dips, since these include up to 3 insertions in the gcode each.

delawarehunter, 2019-06-09:
Ok sounds great.  I'll take a look at the gcode and see if I see anything specific that stands out.  I appreciate the help and I'll let you know if I make any progress on the integrated version.

domesticatedviking, 2019-06-09:
I think it is fairly likely that I will eventually decide that maintaining builds for the integrated version is less time consuming than keeping the PP script going lol.   Currently downloading Xcode to build the mac version.

delawarehunter, 2019-06-09:
haha, yea I hope it's easier.  I like the idea of having it fully integrated.

domesticatedviking, 2019-06-09:
It is just so much more satisfying to setup the integrated version for a print.  Hopefully it works.   The gcode it generates looks good but I'm actually setting up for my first test print with it now:
[image]

delawarehunter, 2019-06-10:
I've been trying to work on getting your version of the PrusaSlicer to build, but it's just not working for some reason.  I get 8 errors and alot of warnings and it says the build failed.  The fatal error has something to do with not being able to open a libslic3r.lib file.  Worked on it 2 nights in a row, I just don't understand much about it.

domesticatedviking, 2019-06-10:
Yeah, you just described my last 3 nights.   Planning to keep snapshots of the VM's I'm building with so that I never have to go through this process again once I get it working.

domesticatedviking, 2019-06-10:
Oh, and if you are still trying, be sure to grab the latest commits - my test revealed an error in the generated gcode which was causing the toolchange temperatures not to be applied at all.   I'm also planning to add a maximum part fan speed field for the toolchange temperature section as this was not working well with the hotend PID routine.

delawarehunter, 2019-06-14:
So, I was able to finally get your version of the PrusaSlicer to build on Windows.  When I look at the GCode, how do I verify that the skinnydip code is in the gcode?  I don't see the spot where it tells me how many tool changes and how many dips were added.

domesticatedviking, 2019-06-27:
If you run 64 bit windows, check out this early [build of PrusaSlicer with Skinnydip routines built in](https://mega.nz/#!b3I3kYBT!ybRMhsaUdL1NJUiPGQE2U3SMaEWA6oQwC1-xLjuixLQ) --
Skinnydip settings are in Filament Settings > Advanced (same place you set up cooling moves and ramming) - you will need to click the advanced or expert button in the top right corner to see them
It's an alpha so there are likely to be a few issues yet. It would be a good idea to back up your Slic3r profile directories before you run it.

delawarehunter, 2019-06-29:
I'll give it a shot.  Just so you know, for some reason, I get an error when trying to load .3mf files in your version.  These are files that I got from Prusa for their new MMU2S enclosure.

domesticatedviking, 2019-06-30:
> I'll give it a shot. Just so you know, for some reason, I get an error when trying to load .3mf files in your version. These are files that I got from Prusa for their new MMU2S enclosure.

edit: It didn't work in the version I was able to get compiled or the one that you linked above.

I think that may be an issue with the official build as well from what I've been reading.

domesticatedviking, 2019-07-03:
> So, I was able to finally get your version of the PrusaSlicer to build on Windows. When I look at the GCode, how do I verify that the skinnydip code is in the gcode? I don't see the spot where it tells me how many tool changes and how many dips were added.

If you use the integrated version, there will be no skinnydip header.  There will also be no stats.   You can open the gcode file and search for "skinnydip"  and you should find the relevant gcode.