domesticatedviking/skinnydip #10: Skinnydip not being executed on the "last Filament"

germanyxx3, 2019-06-20:
Hello,
I finally recieved my Torlon tube and I'm using Skinnydip with it.
David Shaely and you are my saviors! Thank you for creating this masterpiece that Prusa didn't came up with!

But Skinnydip isn't working 100% right for me
I'm still using 1.0.3 because it's the only version that seems to work somehwat for me. I'm using 3 different Filaments and everytime the last highest extruder number does not get dips added.
(In my case it's extruder 3)
I also tried shifting all 3 colors one number up (so Extruder 2-4 are assigned) but that didn't change anything.
Whenever I use 1.0.5 Not a Single dip is added even though the only thing I changed was replacing the old Skinnydip.py 1.0.3 with the new 1.0.5 Skinnyidp.py

Here is the link to the non-Skinnydip, with Skinnydip and log files for both 1.0.3 and 1.0.5 
The print is 3 2x2x2cm cubes not tuching each other. But every cube has assigned its own color
https://www.amazon.de/clouddrive/share/ifevlbGCic2iLnhaPIPk2aMIw7GO3SlZhYnaZDlqJsp

I also tried assigning the same Filament type on all 5 extruders like delawarehunter  https://github.com/domesticatedviking/skinnydip/issues/9#issuecomment-500175614 did it.
Before I read that post I had two different Filament types assigned but the amount of dips didn't change.

I'm using Slic3r 1.41.3+win64

domesticatedviking, 2019-06-20:
Thanks for the detailed report.  Unfortunately, the pattern-search approach used by this tool is not 100% reliable for all files. It was good enough to function as a proof of concept, but I am shifting focus to my fork of PrusaSlicer that has the SkinnyDip routines built in, because it is FAR easier to configure and 1000X more reliable.   The source code for my fork is already up (see [here](https://github.com/domesticatedviking/PrusaSlicer/tree/skinnydip-integrated/))   -- with any luck I should have binaries available for Windows, Mac, and Linux within the next week or two.

germanyxx3, 2019-06-20:
I guess you will post an update on github once it's available?
Then I'll switch to Prusaslicer. Right now I see no point in doing so. Slic3r is working (besides that issue) flawless.
I can wait another two weeks. I've already invested more than a month in researching for mmu2s fixes so that time doesn't really make matters worse. I'm already thankful that you are doing that!

domesticatedviking, 2019-06-22:
> Then I'll switch to Prusaslicer. Right now I see no point in doing so. Slic3r is working (besides that issue) flawless.
The only difference between PrusaSlicer and Slic3r is the name and version number.  They are the exact same product otherwise.

domesticatedviking, 2019-06-27:
If you run 64 bit windows, check out this early [build of PrusaSlicer with Skinnydip routines built in](https://mega.nz/#!b3I3kYBT!ybRMhsaUdL1NJUiPGQE2U3SMaEWA6oQwC1-xLjuixLQ) -- 
Skinnydip settings are in Filament Settings > Advanced (same place you set up cooling moves and ramming) - you will need to click the advanced or expert button in the top right corner to see them
It's an alpha so there are likely to be a few issues yet.  It would be a good idea to back up your Slic3r profile directories before you run it.

germanyxx3, 2019-07-21:
I'll try it out once I get back into 3d printing.
Right now I need a break. Finding and fixing the issues of my mmu2 did cost me too much energy.